hybridgroup/gort #28: Help Installing Gort (Ubuntu 64bits)

nicolasgaraza, 2015-01-03:
I'm installing Gort on Ubuntu 64 bits but I have the following error after installing using the gort_0.3.0_amd64 file:

dpkg: error processing archive /home/nicolas/Descargas/gort_0.3.0_i386.deb (--install):
 parsing file '/var/lib/dpkg/tmp.ci/control' near line 7 package 'gort':
 blank line in value of field 'Description'

Thanks!!

pcostesi, 2015-01-09:
I've got the same issue with armhf:

```
pcostesi@rpiaplus1 ~ $ sudo dpkg --install gort_0.3.0_armhf.deb 
dpkg: error processing gort_0.3.0_armhf.deb (--install):
 parsing file '/var/lib/dpkg/tmp.ci/control' near line 7 package 'gort':
 blank line in value of field 'Description'
Errors were encountered while processing:
 gort_0.3.0_armhf.deb
```
The contents of the `control` file are:
```
Package: gort
Priority: extra
Maintainer: unknown <example@example.org>
Section: devel
Version: 0.3.0
Architecture: armhf
Description: ?
 
```
(Note the blank line)

Removing the blank line solved the problem in Raspbian. A new build should fix this. Meanwhile, [here](http://askubuntu.com/a/191854/467) is an easy way to edit the `deb` file (as a workaround).

I hope this helps.

gorhgorh, 2015-01-18:
i can confirm the bug still stands as of today :) thanks for the fix @pcostesi

elhowm, 2015-01-20:
"I can confirm the bug still stands as of today" +1

dkinzer, 2015-02-04:
I installed gort on ubuntu 14.04 LTS 64 without issue by following the instructions in the README of this repository.

hasbisevinc, 2015-02-11:
@dkinzer  I read README file. but still, I cant install gort. I am using ubuntu 14.04 LTS - 64bits
any tips?

dkinzer, 2015-02-12:
@hasbisevinc did you make sure you get follwing?:

```
go get github.com/jteeuwen/go-bindata/...
```

alioygur, 2015-04-19:
i have same problem.

graveto, 2015-05-20:
I have used the link for editing .deb files provided above by pcostesi. There is an extra empty line at the end of the file. After removing that line and adding random description string, I was able to install successfully. 

File before:
  1 Package: gort
  2 Priority: extra
  3 Maintainer: unknown <example@example.org>
  4 Section: devel
  5 Version: 0.3.0
  6 Architecture: amd64
  7 Description: ?
  8 .

File after:
  1 Package: gort
  2 Priority: extra
  3 Maintainer: unknown <example@example.org>
  4 Section: devel
  5 Version: 0.3.0
  6 Architecture: amd64
  7 Description: "Some robot stuff"

Gort is working now.

This should be easy to fix.

deadprogram, 2016-04-05:
Now that we are running much newer version of the various build tools, is this still a problem for anyone?

deadprogram, 2016-05-26:
The .deb file is now correctly being built by goxc since the last release. Closing, please reopen if anyone has any further trouble.

closed 2016-05-26 by deadprogram