hybridgroup/gort #29: Uploading hex file does not work for arduino gobot hexfile.

dkinzer, 2015-02-04:
I'm not sure if this is a gort issue or a gobot issue, but trying to upload a compiled example code to my arduino device fails with below output:

```
> gort arduino upload blinker /dev/ttyACM0

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "blinker"
avrdude: invalid record at line 5991 of "blinker"
avrdude: read from file 'blinker' failed

avrdude done.  Thank you.

2015/02/04 10:33:52 exit status 1
```

Note that just running the file directly works, but then if I unplug the USB cord from the device, the program stops working.

vickash, 2015-02-08:
Are you using an Arduino model other than the UNO?

dkinzer, 2015-02-09:
No, I'm using only Arduino Uno.

zankich, 2015-02-09:
@dkinzer Are you trying to upload a compiled gobot program to the arduino? Unfortunately that won't work because gobot uses [firmata](https://github.com/firmata/arduino) to remotely control the arduino from a host computer. The `gort arduino upload` command is intended to upload an arduino `sketch` program, not go binaries.

dkinzer, 2015-02-09:
@zankich thanks.   So I guess if I want to write for something that will not be connected directly to some host, then I'm stuck with sketch or C or C++?

closed 2015-02-09 by dkinzer

zankich, 2015-02-09:
@dkinzer yes unfortunately those are your options for programming an arduino directly.

dkinzer, 2015-02-09:
For posterity's sake I want to note that if one is only using gort to upload sketches then they should probably be using [Ino](https://github.com/amperka/ino) instead.

jy03189211, 2015-09-24:
@zankich  I happened to have the same situation(I was trying to upload the compiled binary file), noticed that I should upload the firmata. How should I use my go code to with firmata? What if my have multiple go files in one folder?

deadprogram, 2015-09-24:
@dkinzer I like everything about Ino except adding additional dependencies aka Python. I like Python fine, but not dependencies. Agree that Gort is not intended as a general purpose Arduino solution. For sure, however, Gort can do a better job at handling different .hex files and platforms, and we're working on that.

@jy03189211 as far as how to use Gobot with Firmata, please check out https://github.com/hybridgroup/gobot/tree/master/platforms/firmata if you have not seen it yet.

jy03189211, 2015-09-24:
@deadprogram on the page, there are only how to install the dependency, but there's nothing explaining how to use the code. should I just "go run file.go" or "gort arduino upload firmata /dev/serial " if the second is right, how can I define which code file to run?

deadprogram, 2015-09-24:
Gobot does not run on the Arduino itself. It uses the Firmata serial protocol to communicate from your Go program running on your computer, to a connected Arduino that is running the Firmata C++ sketch on the Arduino itself.

Once you have uploaded the Firmata to the Arduino, you can send the Arduino commands from your computer. It is not possible currently to run Golang on the Arduino itself.

Hope that helps!

jy03189211, 2015-09-24:
@deadprogram Thanks for the information:) that helps me understand it.
Now the example code is working. But every time I press Ctrl-c to stop the program... It stuck in the shell. Any idea?
Plus I tried the "gobot generate" command to make a project dir and I run the main.go file. It gives error "can't find import: "github.com/hybridgroup/gobot" while the example code from the official website containing the same dependency works ...

deadprogram, 2015-09-24:
That usually means your GOPATH is not set correctly relative to code you are trying to run.