hybridgroup/gort #31: Uploading firmata to arduino: nothing happens (OS X)

droptheplot, 2015-02-05:
Trying to upload firmata to arduino with
```
sudo gort arduino upload firmata /dev/tty.usbmodem1421
```
and nothing happens. How do I know what's wrong?

* OS X Yosemite 10.10.2
* Port works fine with Arduino IDE

zankich, 2015-02-09:
@droptheplot did you first run `gort arduino install`? That installs the required build tools for programming an arduino.

droptheplot, 2015-02-11:
@zankich yes.
```
Warning: avrdude-6.1 already installed
```

shemerey, 2015-03-06:
same for me, just getting 

    avrdude: stk500_recv(): programmer is not responding
    avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x3c

shemerey, 2015-03-06:
So it looks like i've managed to fix it, based on a source code we have "hardcoded" `avrdude` command, with fixed version etc. so I've put on debug mode in Arduino IDE and get correct command for `avrdude`, just replace hex file

JesterXL, 2015-08-27:
I'm running into this problem with a [Ringo](http://www.plumgeek.com/). Is there something I can do to help?

deadprogram, 2015-08-31:
What needs to happen, is that this line of code needs to be able to take a parameter corresponding to which target microcontroller is in use:

https://github.com/hybridgroup/gort/blob/master/commands/arduino.go#L105

I will try to get to this, but would be delighted to accept a PR if someone else can do it first :)

netbrain, 2015-09-01:
My fix might not be relevant for all, but this atleast fixed my issue with my arduino nano.

deadprogram, 2016-02-17:
The new 0.5.2 release of Gort is now out, and should address this issue. Please reopen if you need further assistance, or open a new issue. Thanks!

closed 2016-02-17 by deadprogram