hybridgroup/gort #32: avrdude CLI parameters for different Arduino products

vickash, 2015-02-09:
I'm not a Go developer, but a quick look at the source tells me `gort` Arduino support is limited to the UNO or other boards with the ATmega328 chip. This covers the popular basic boards, but leaves out everything else.

The `avrdude` options needed to make this work are `-c`, which tells `avrdude` what programmer to use and `-p`, which indicates the part number of the chip on the board.

It would be great if there were an option in `gort` like `-board mega2560`, which would let the user refer to the board by product name rather than the chip. `gort` would parse that option and then give the correct parameters to `avrdude`.

To that end, I'm posting all the combinations I know to be working here. This covers basically the entire product line from the Diecimila (2007) onward, except for the Due, which is ARM-based and requires `bossac` instead of `avrdude`. That's another issue.

Additionally the **`-b` option should be left out**. `avrdude` varies the default baud based on the chip, and I've yet to see an error solved by explicitly setting it.

## The Parameters

**UNO / Ethernet / Nano v3.0 / Mini v5 / Late Duemilanove / Any with ATmega328/328p chip:**
````
-c arduino -p m328p
````

**Nano v2.3 or earlier / Mini v4 or earlier aka Stamp / Early Duemilanove / Diecimila / Any with ATmega168 chip:**
````
-c arduino -p m168
````

**Mega aka Mega1280 (Old Model) / Any with ATmega1280 chip:**
````
-c arduino -p m1280
````

**Mega2560 (Current Model) / Mega ADK / Any with ATmega2560 chip:**
````
-c stk500v2 -p m2560
````

**Leonardo / Robot / Esplora / Micro / Yun / Any with ATmega32u4 chip:**
````
-c avr109 -p atmega32u4
````

#### Baud Reset for 32u4-based Boards

Boards based on the 32u4 require the system to set the baud rate to 1200 prior to running `avrdude`.

On OS X: `stty -f /dev/tty.whatever 1200`
On Linux: `stty -F /dev/tty.whatever 1200`
Not sure how to do this on Windows.

This is an odd quirk in that, once you run the command once, `avrdude` will run successfully on that device until the machine reboots (even across disconnects and reconnects). However, it does no harm to run it prior to each call to avrdude, which is what I have done in the past.

## Possible Implementation

That's a lot of boards, and there's overlap between the m168 and m328 for the older products.

Since m328p is the most popular, and m168 is no longer built into new boards, `gort` should default to the m328, requiring users of older boards with a m168, to know this and specify so. All other boards would need to be specified explicitly.

The `gort` options could look something like this:
````
-board m328 | uno | nano | mini | ethernet (default option)
-board m168 | diecimila | stamp
-board mega | mega1280
-board mega2560 | megaADK
-board leonardo | robot | micro | esplora
````

All the parameter values on a specific line would produce the exact same output parameters for `avrdude`, corresponding with my findings above, but could be accessed by different names for ease of use.

Lastly, the Mini and Stamp have never had USB ports built in. They were intended to be programmed via a USB-to-serial adapter, then disconnected and run standalone. Since this doesn't fit with Artoo/Cylon/Gobot, it's debatable whether they should be included. But if `gort` is designed to be a general-purpose flashing utility, I think they should.

zankich, 2015-02-09:
@vickash I think it's a great idea to expand the capabilities of which architectures avrdude can compile to. I only have an uno, so I haven't tested with the other boards, but it should be fairly easy to expose the programmer via a command line flag.

vickash, 2015-02-09:
I have a lot of these covered hardware-wise, so I can help with testing.
What I mentioned above is strictly upload, not compilation.

I also started work yesterday on a script that leverages the newest ide
(1.5.8) to compile different binaries for each product from a single sketch
file. Should put that up in a few days.

I have standard firmata compiling successfully on all the official arduino
products and dumping neatly named hex files to a single folder. Just the
interface needs some work.

shemerey, 2015-03-10:
@vickash  :+1:  that can solve following issue https://github.com/hybridgroup/gort/issues/31

vickash, 2015-03-14:
Arduino batch compile script:
https://github.com/vickash/abc

th3m4ri0, 2015-05-30:
:+1:. Can't use with my Arduino Leonardo for the moment.

rafmagana, 2015-05-31:
@th3m4ri0 we're aware of this problem and we have plans to fix it in the near future, we'll let you know when it's fixed already, sorry for inconvenience.

th3m4ri0, 2015-06-01:
No problem, thanks for the update :)

pires, 2015-09-19:
Leonardo support would be awesome for me to work with littleBits Arduino bit.

deadprogram, 2016-02-17:
The new 0.5.2 release of Gort is now out, and should address this issue. Thank you SO much to @vickash for all the details.

deadprogram, 2016-04-05:
Now that this has been released for a while, I am closing this issue. Thanks again, everyone.

closed 2016-04-05 by deadprogram