hybridgroup/gort #35: `gort serial` error uploading Firmata to Intel Edison

aelkus, 2015-07-21:
I am using OS X 10.10.4, [Intel Edison with Arduino breakout](https://www.sparkfun.com/products/13097), and gort_0.3.0_darwin_amd64. Here is the output received going through [this tutorial](http://gobot.io/documentation/platforms/arduino/). 

```
$ gort arduino upload firmata /dev/cu.usbserial-AJ035DL1 
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20

avrdude done.  Thank you.

2015/07/21 15:56:31 exit status 1
```
I have been able to upload modified Blink sketches using the Arduino IDE. But when I attempted to upload the Standard Firmata sketch from Examples, I received this output: 

```
/Applications/ArduinoEdison.app/Contents/Java/libraries/Firmata/utility/EthernetClientStream.cpp:71:22: error: conversion from 'IPAddress' to 'unsigned int' is ambiguous
   if (this->localip!=localip)
```
Lastly, using Gort caused the device names for Edison/Arduino breakout to change. The Blink sketches were uploaded using `/dev/cu.usbmodem1d113` which changed once Gort was run to `/dev/cu.usbserial-AJ035DL1`. At this point not even Blink sketches via Arduino IDE upload. 

Any insight appreciated.

deadprogram, 2015-09-01:
I have never tried to run Firmata on the Edison. Instead, I use the Edison's GPIO/I2C interfaces via either Cylon.js or Gobot. Not sure if the Firmata sketch will even run on Edison?

deadprogram, 2017-01-04:
Closing due to age of issue, please reopen if needed. Thanks!

closed 2017-01-04 by deadprogram