hybridgroup/gort #38: gort arduino upload error

yremmet, 2015-09-13:
When I run `gort arduino upload /dev/cu.usbserial-A800evvq` it aborts with an error.

```
panic: runtime error: index out of range

goroutine 16 [running]:
runtime.panic(0x2f75c0, 0x51f35c)
	/home/adrian/code/go/src/pkg/runtime/panic.c:279 +0xf5
github.com/hybridgroup/gort/commands.func·002(0x2085ee240)
	/home/adrian/code/go/gopath/src/github.com/hybridgroup/gort/commands/arduino.go:90 +0x1996
github.com/codegangsta/cli.Command.Run(0x33d590, 0x7, 0x0, 0x0, 0x3a0090, 0x35, 0x0, 0x0, 0x0, 0x0, ...)
	/home/adrian/code/go/gopath/src/github.com/codegangsta/cli/command.go:101 +0xc6f
github.com/codegangsta/cli.(*App).Run(0x2085ee0c0, 0x2085a6000, 0x4, 0x4, 0x0, 0x0)
	/home/adrian/code/go/gopath/src/github.com/codegangsta/cli/app.go:130 +0x953
main.main()
	/home/adrian/code/go/gopath/src/github.com/hybridgroup/gort/gort.go:25 +0x2b5

goroutine 17 [runnable]:
runtime.MHeap_Scavenger()
	/home/adrian/code/go/src/pkg/runtime/mheap.c:507
runtime.goexit()
	/home/adrian/code/go/src/pkg/runtime/proc.c:1445

goroutine 18 [runnable]:
bgsweep()
	/home/adrian/code/go/src/pkg/runtime/mgc0.c:1976
runtime.goexit()
	/home/adrian/code/go/src/pkg/runtime/proc.c:1445

goroutine 19 [runnable]:
runfinq()
	/home/adrian/code/go/src/pkg/runtime/mgc0.c:2606
runtime.goexit()
	/home/adrian/code/go/src/pkg/runtime/proc.c:1445
```

I'm using OS X 10.10.5
gort version: 0.3.0 darwin_amd64
avrdude verison: 6.1 (installed via homebrew)
and an Arduino Duemilanove

Since these are my first attempts with gort I'm at a loss.
What is wrong?

alexandru-calinoiu, 2015-10-02:
got the same problem running ubuntu 14.04 and gort version 0.3.0

X1ting, 2015-12-19:
have the same problem on ubuntu 14.04 and gort version 0.4.1

deadprogram, 2016-02-17:
Hi, everyone

The command should be:

```
gort arduino upload firmata /dev/cu.usbserial-A800evvq
```

Note the missing param. We should do a test to verify the correct number of minimum params, to avoid such errors.

deadprogram, 2017-01-04:
This was fixed by @lcguida now closing. Thanks everyone!

closed 2017-01-04 by deadprogram