hybridgroup/gort #39: Limited options when running Gort

kenmarold, 2015-11-12:
I'm trying to run Gort on OSX 10.11 and I'm getting a very limited selection of options, specifically only 'help' and 'version'. When I run gort -V in Terminal I get 0.0.0 which doesn't seem right and gort -h just yields the same options below. What's going on here? I installed Gort with npm and it didn't give me any errors.

 Usage: gort [options]

  Options:

    -h, --help     output usage information
    -V, --version  output the version number

deadprogram, 2015-11-12:
npm? That would be a different "gort". 

This repo is for Gort which is written in Golang. If you want to install the latest version, download the binary for your OS from http://gort.io/documentation/getting_started/downloads/

Hope that helps!

kenmarold, 2015-11-12:
Okay, I download the binary and run it, I get a terminal window with this info, but then I can’t do anything else, what do I need to do next?

Last login: Thu Nov 12 13:06:33 on ttys001
d-ip-10-194-75-201:~ kenmarold$ /Users/kenmarold/Downloads/gort_0.3.0_darwin_amd64\ 2/gort ; exit;
NAME:
   gort - Command Line Utility for RobotOps

USAGE:
   gort [global options] command [command options] [arguments...]

VERSION:
   0.3.0

COMMANDS:
   scan		Scan for connected devices on Serial, USB, or Bluetooth ports
   bluetooth	Pair, unpair & connect to bluetooth devices.
   arduino	Install avrdude, and upload HEX files to your Arduino
   spark	Upload sketches to your Spark
   digispark	Configure your Digispark microcontroller
   crazyflie	Configure your Crazyflie
   klaatu	barada nikto
   dronedrop	Install, uninstall, update and download dronedrop firmware
   help, h	Shows a list of commands or help for one command
   
GLOBAL OPTIONS:
   --help, -h		show help
   --version, -v	print the version
   
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

Best,
Ken

Ken Marold, AIA Associate // AIGA // SEGD
LOCOMOKU // Founder + Principal // 405-365-7177
www.locomoku.com <http://www.locomoku.com/>
twitter.com/locomokudesign <https://twitter.com/locomokudesign>

deadprogram, 2015-11-12:
@kenmarold you use Gort from the Terminal, not from the Mac Finder. First, extract the executable from the ZIP file. Next, open a new Terminal window and type this command, and hit enter:

`cd /whereever/you/put/gort`

substituting the correct directory wherever you extracted the `gort` file.

Last, type this command into the terminal window, and hit enter:

`./gort`

Hope that helps!

kenmarold, 2015-11-12:
that did it. thank you!! i was just typing gort in the directory that I put gort instead of ./gort.

deadprogram, 2015-11-12:
OK, great. Glad to help!

closed 2015-11-12 by deadprogram