hybridgroup/gort #42: gort arduino firmata relay on by default

JesusIslam, 2015-11-21:
After I uploaded the firmata firmware to my UNO, everytime I power it up, the NC relay always got turned on immidiately. 

How can I make it so it won't turned on immidiately (off by default)? Should I make custom firmata firmware? If so, how do I do that?

Thanks.

deadprogram, 2015-11-22:
Hi @JesusIslam 

When the Arduino is first booting up before the firmware loads, you cannot count on the pins being in a consistent state. Normally the circuit you would wire up would accommodate this. For example:

[image]

From this useful post:

http://www.electroschematics.com/8975/arduino-control-relay/

Hope that helps!

deadprogram, 2016-02-17:
Hi, @JesusIslam did you get your problem resolved? If you still need further assistance, please reopen this issue. Thanks!

closed 2016-02-17 by deadprogram