hybridgroup/gort #45: Can't download .deb packages

jullit31, 2016-05-09:
I only get something like this, when trying to download any of the .deb packages:

````
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>NoSuchKey</Code>
<Message>The specified key does not exist.</Message>
<Key>0.6.1/gort_0.6.1_amd64.deb</Key>
<RequestId>A2BE481AA5274077</RequestId>
<HostId>
oOjZGRSPQ5GExI19da3QGjFfPD9zxf6VYyJMarj0nUA7Jdtqd2REtyCAt8wIyumaBKaPelaLw/A=
</HostId>
</Error>
````
Other downloads work.

deadprogram, 2016-05-12:
Hi @jullit31 

Thanks for reporting this. There was an error in our build for the last release. We've now updated to 0.6.2 which should correct your problem. Please give it a try.

jullit31, 2016-05-13:
Download works now.
Thanks.

closed 2016-05-13 by jullit31