hybridgroup/gort #46: There are lots of versions of firmata out there, which one is it that gort uploads?

troywweber7, 2016-09-08:


deadprogram, 2016-09-08:
Hi @troywweber7 it is StandardFirmata.

Hope that helps!

troywweber7, 2016-09-08:
Thank you, that resolves it @deadprogram.

Additionally I'm looking to extend Cylon functionality to include the ability to drive stepper motors. I have a sense that this may involve me adding support for other firmatas on cylon-firmata or at the very least extending the cylon-gpio module. I'm seeking direction on one or both of those tasks [here](https://github.com/hybridgroup/cylon-firmata/issues/47) and [here](https://github.com/hybridgroup/cylon-gpio/issues/55). I apologize for the duplicates. I might also be interested to help add support to gort to easily upload alternative firmatas.

closed 2016-09-08 by troywweber7

deadprogram, 2016-09-10:
Hi @troywweber7 a contribution that allowed for uploading other Firmata versions would be greatly appreciated.