hybridgroup/gort #47: Mindwave Mobile on Ubuntu 16.04 throws error when trying to connect

Talamantez, 2016-09-28:
Hi - I'm connecting a Mindwave Mobile to Ubuntu via a bluetooth dongle. I switch the device to 'on/pair', and then run: 

hcitool scan 
expected output: 
Scanning ...
	20:68:9D:4C:0D:B7	MindWave Mobile

gort scan bluetooth
expected output:
Scanning ...
	20:68:9D:4C:0D:B7	MindWave Mobile

gort bluetooth connect 20:68:9D:4C:0D:B7
error output:
2016/09/27 23:40:52 exec: "bluez-test-serial": executable file not found in $PATH

I've looked around a bit and haven't discovered the apt repo for that tool.  Any suggestions?

Cheers,
Robert

deadprogram, 2016-10-04:
Hi @Talamantez seems like you might have an older version of Gort installed? We switched to using `hciool` directly a couple of versions ago. Can you please run:

```
gort --v
```

And provide the output? Thanks.

Talamantez, 2016-10-10:
gort version 0.6.2 - thanks @deadprogram

Talamantez, 2016-10-10:
hi - I switched platforms, I'm going to close this issue - thanks!

closed 2016-10-10 by Talamantez