hybridgroup/gort #48: Cannot connect to BB-8 Sphero. Beleive it is a security level issue and/or random address

rfestag, 2016-12-04:
I am trying to use artoo (ruby library for interfacing with various projects, including Sphero). Their documentation suggests using gort to pair/connect to the device, which (as I understand) should create a /dev/rfcomm0 device for artoo to connect to. Their documentation appears to be out of date (since it tells you to pair, but there doesn't appear to be a pair subcommand anymore). In any case, when I try to simply connect as follows, I get "Cannot create connection. Input/output error":

`sudo gort bluetooth connect <ADDR> hci0`

I found that using gatttool, I can connect only with I set my address to random and security level to medium.

`gatttool -l medium -t random <ADDR> -I`

This lets me connect to the BB-8 (although I haven't tried doing more than printing characteristics because I'm not familiar enough with the system), but it doesn't create a /dev/rfcomm0 (or other device) I can point artoo at. 

Is there something I'm missing? I assume there currently is no way to set the security level, and I assume it defaults to low. Is that correct? If so, what would it take to support other security levels and/or random addresses?

deadprogram, 2016-12-04:
Hi @rfestag to connect to a BB-8 you need to use Bluetooth Low Energy aka BLE aka Bluetooth 4.0, not Bluetooth Classic aka Bluetooth 2.0. With BLE there is not a pairing step, you do not need to use Gort for this.

Check out the Gobot Ollie example, the Sphero Ollie uses the same API at the Sphero BB-8: https://github.com/hybridgroup/gobot/blob/master/examples/ollie.go

Hope that helps!

rfestag, 2016-12-04:
Thanks for the quick reply. If I'm understanding correctly, that implies that BB-8 would require a new adapter in artoo if I wanted to use Ruby to control it? Since the sphero adapter expects a device as a parameter, and connecting the same way that example shows directly connects without allocating a device?

deadprogram, 2016-12-04:
Yes you are correct about needed to add the BLE support to Artoo.

deadprogram, 2017-01-04:
The new version of Gort allows for BLE scanning on Linux by using the `gort scan ble` command.

Now closing this issue, thanks again for the feedback.

closed 2017-01-04 by deadprogram