hybridgroup/gort #49: Install CHIP pwm using device tree overlays

deadprogram, 2017-03-18:
Add commands to be able to install CHIP pwm using device tree overlays.

erkkah, 2017-03-19:
Looking through the list of current gort commands, it seems like they all run "remotely" to control connected devices. Is that the overall intention or would it be a valid use case to run gort directly on the device?

deadprogram, 2017-03-20:
Some run remotely, some run against the local system, depending on the use case.

In the case of the C.H.I.P. device tree overlay, it would need to run directly on the system you want to configure.

erkkah, 2017-03-20:
I see.
The implementation would only work when run on the C.H.I.P. and on a stock system image. Hopefully the system init script wars will not cause too much troubles.
I'll give it a shot when I have the time, and see how it turns out.

The implementation still depends on a modified `dtc` being installed, maybe that should be a separate command?

deadprogram, 2017-03-20:
Sure, sounds good.

erkkah, 2017-03-22:
Hey, @deadprogram, I have a working C.H.I.P implementation on my fork now: https://github.com/erkkah/gort/commit/f0a4ad639bfd37eb528737be6b801c6b257e4bed

Did not want to create a PR since there is no dev branch.

deadprogram, 2017-03-23:
Hi @erkkah sorry about that, you are correct. Not sure why that remote branch was deleted, but it has now been recreated.

deadprogram, 2017-03-23:
This has been addressed thanks to @erkkah with https://github.com/hybridgroup/gort/pull/50

I will close once a new version of Gort has landed with this code.

deadprogram, 2017-03-30:
The version of Gort with this has been released. Closing this issue now. Thanks again, @erkkah

closed 2017-03-30 by deadprogram