hybridgroup/gort #51: Error when I try brew install hybridgroup/tools/gort

Killians06, 2017-04-06:
Hi ! Like what I said in tittle, when I try to install Sort, i've an error on Mac OS X Sierra:

```
Error: SHA256 mismatch
Expected: be09312d577c477d9720b591044773a8f4f95655c3112a8ab901c5de17fbd4ff
Actual: acc7896f3765fbaea13f2007d87e7914af06d05d681a5ebad4fa22724e560b43
Archive: Users/**hidden**/Library/Caches/Homebrew/gort-0.8.1.zip
To retry an incomplete download, remove the file above.
```

What can I do for fix that ?

Cheers,

deadprogram, 2017-04-06:
Hi @Killians06 please give another try. The last release updated the SHA but not the filename. This commit should correct: https://github.com/hybridgroup/homebrew-tools/commit/92534e3111f7d8d839173ec9fa1d6244d1d119b2

Killians06, 2017-04-06:
That's good !
Thank you for your last commit !

closed 2017-04-06 by Killians06

mattruzicka, 2019-05-10:
I'm seeing this issue on macOS Mojave

Stefanbracke, 2019-09-30:
Is there a solution to this problem?