hybridgroup/gort #53: Wrong URL is refenced in Win64 build of gort 0.8.1

JDogHerman, 2017-06-13:
When attempting to run a command for `gort digispark` the zip is attempted to be downloaded from littlewire.cc rather than S3 mirror. (fix was applied in commit: https://github.com/hybridgroup/gort/commit/a29178c001cc932beced24b953100cdd4854162d)

```
gort.exe digispark install
Gort lives
Downloading http://littlewire.cc/resources/LittleWirev13Install-Win.zip to LittleWirev13Install-Win.zip
7204 bytes downloaded.
2017/06/13 17:03:48 zip: not a valid zip file
```

The build I downloaded was referenced on this page http://gort.io/documentation/getting_started/downloads/ 
and the link was:
https://s3.amazonaws.com/gort-io/0.8.1/gort_0.8.1_windows_amd64.zip

This file's MD5 hash of the build file with this issue is  57484C4009D044DC7A181B85225CE7DB

deadprogram, 2017-06-14:
Hi @JDogHerman thanks for the report. Yes, I was aware of this, and in fact we have a fix coming in the next version: https://github.com/hybridgroup/gort/commit/8922c3b3326933bb07b2d8a9bc9802abedae03ca

We should be putting out a new release any day now, thanks again for reporting.

deadprogram, 2017-06-15:
The new version of Gort was released. Now closing, reopen if needed. Thanks again for reporting, @JDogHerman

closed 2017-06-15 by deadprogram