hybridgroup/gort #55: Documentation: gort digispark install does more than just downloading the software

christophberger, 2017-07-20:
Hi, 

Minor issue.

gort's help text says:

```
gort digispark install # installs software to upload firmware to Digispark
gort digispark upload [littlewire] # uploads firmware to Digispark
```

but actually `go digispark install` does both steps in one: install software to upload firmware to Digispark, AND upload firmware (littlewire) to Digispark.

So the actual behavior is not in line with the help text.

deadprogram, 2017-07-25:
Hi @christophberger you are indeed correct, seems like this line of code https://github.com/hybridgroup/gort/blob/master/commands/digispark.go#L42 should be removed.

Want to create a PR with that change? :)

christophberger, 2017-07-26:
Sure. Shall I create the PR against the dev branch, since the fix *does* change the behavior?

deadprogram, 2017-07-26:
Yes, please, and thank you!

christophberger, 2017-07-26:
Looks like dev is behind master. digispark.go still contains the old invalid urls to littlewire.cc on dev. What's the best way to move forward?

deadprogram, 2017-07-26:
I just rebased master into dev, so you should be good to go now. Sorry about that.

christophberger, 2017-07-26:
No problem. I created a PR just now.

deadprogram, 2017-07-26:
Thanks for the PR. I will leave the issue open until the next release of Gort.