hybridgroup/gort #64: Got a SHA256 mismatch Error when I try 'brew install hybridgroup/tools/gort'

ruanqisevik, 2019-07-03:
Hi ! I've got an error on Mac OS X High Sierra 10.13.6 when I tried to install gort with brew:
```
Error: An exception occurred within a child process:
  ChecksumMismatchError: SHA256 mismatch
Expected: 744fb23c83b1c5ac8612f08c3c9f791ad12a2bef0a9d0b79eb695da6a124025b
  Actual: 05b1baacdc0b4b6202a01b0e622d81ba60d2ed1963e6f14563f8caafc4ad8460
 Archive: /Users/roy/Library/Caches/Homebrew/downloads/87962709ef2f694dbd2349834d7d1ec4bba1c2a036bb16a4ee723eda0d067e24--gort_0.9.0_darwin_amd64.zip
To retry an incomplete download, remove the file above.
```
I have read about issue #51;
I think maybe there is the same problem;

Look forward to your reply！

Stefanbracke, 2019-09-30:
Any way we can do this ourself?

Stefanbracke, 2019-09-30:
Yes we can: brew edit hybridgroup/tools/gort and replace the wrong SHA256 value with the actual one.

Caldis, 2021-02-20:
Since I found an issue that needs to be fixed, I updated the DMG file after the release resulting in a change to the SHA, so you may have to wait for the cask update